softypit/esp32_mqtt_eq3 #1: Missing mttq.h

Stat1c-V01D, 2018-03-18:
eq3_wifi.c:31:18: fatal error: mqtt.h: No such file or directory
compilation terminated.

Compiling with: MySys (minGW32) and ESP-IDF
Already tried mqtt_client.h from the espmqtt Repository with no luck

grzegorz189, 2018-04-08:
eq3_wifi.c:31:18: fatal error: mqtt.h: No such file or directory
compilation terminated.

Compiling with: MySys (minGW32) and ESP-IDF
Already tried mqtt_client.h from the espmqtt Repository with no luck

tofferlicious, 2018-07-23:
Now having the same issue... I also added the mqtt component to ESP-IDF, but it indeed only provides an mqtt_client.h file. I adjusted the eq3_wifi.c but then I get "unnamed type" error, which is strange...

@softypit could you please check at which revision you cloned the mqtt project from GitHub? I think the mqtt.h was available in an older version. Alternatively or additionally, could you maybe provide a binary e.g. on bintray.com? I would really like to flash it on my ESP32 :)

softypit, 2018-07-23:
Sorry it's been a while since I revisited this repo.
It looks like the tuanpmt/espmqtt api changed completely after I'd cloned the repo. The version I used appears to be from Feb 8th 2018. This contains the mqtt.h required.
You can find it at https://github.com/tuanpmt/espmqtt/tree/fa2f6fa784128c76c33360fbd375ba6c7941e4dc

softypit, 2018-07-23:
I just created a release with the compiled binary included

tofferlicious, 2018-07-24:
@softypit awesome, many thanks :+1: !!!
I was also able to build when using that particular commit from the espmqtt repo.
Could you also check which revision or release-branch you used for the ESP-IDF repo:
https://github.com/espressif/esp-idf

I "fixed" some compatibility issues manually, of which I am not sure, if they could cause misbehavior.

softypit, 2018-07-24:
Now you're making me learn to use GitHub properly :)
Looks like I cloned idf @ commit 2e8441df9eb046b2436981dbaaa442b312f12101.
I also notice the following change I made:

--- a/components/bt/bluedroid/bta/gatt/bta_gattc_act.c
+++ b/components/bt/bluedroid/bta/gatt/bta_gattc_act.c
@@ -1544,7 +1544,7 @@ static void bta_gattc_conn_cback(tGATT_IF gattc_if, BD_ADDR bda, UINT16 conn_id,
     if (p_conn == NULL){
         APPL_TRACE_ERROR("p_conn is NULL in %s\n", __func__);
     }
-
+#ifdef removed // Paul removed this
     if ((transport == BT_TRANSPORT_LE) && (connected == TRUE) && (p_conn != NULL) \
          && (p_conn->service_change_ccc_written == FALSE) && (p_conn->ccc_timer_used == FALSE)) {
         result = bta_gattc_register_service_change_notify(conn_id, bda, &start_ccc_timer);
@@ -1569,7 +1569,7 @@ static void bta_gattc_conn_cback(tGATT_IF gattc_if, BD_ADDR bda, UINT16 conn_id,
                 p_conn->ccc_timer_used = FALSE;
             }
     }
-
+#endif
     bt_bdaddr_t bdaddr;
     bdcpy(bdaddr.address, bda);

IIRC this fixes an issue where the BLE connection didn't clear down properly so wouldn't allow subsequent connections. Suspect this won't be an issue in later IDF versions. If I ever find the time I'll revisit this project and get it working with a specific IDF release.

tofferlicious, 2018-07-25:
> Now you're making me learn to use GitHub properly :)
You're welcome :)

As soon as I have my EQ-3 connected, I could maybe also look into supporting you to adapt this to a current release of IDF. 
Thank you very much again for this project!

Stat1c-V01D, 2018-07-25:
Maybe we all can work together @tofferlicious @softypit  and port this Git to the newest stable/ long-time supported IDF or release a "full package" with the IDF and espmqtt project already included?
Also many thanks from me! It was really annoying to always activate BT on my Phone to change the EQ-3.

Floyddotnet, 2019-02-24:
#14

closed 2019-09-20 by softypit