softypit/esp32_mqtt_eq3 #2: Missing controller.h when building

tofferlicious, 2018-07-22:
First things first, thank you very much for this project!

Unfortunately I am not able to build it though.
I tried the instructions from here:
http://esp-idf.readthedocs.io/en/latest/get-started/index.html#start-a-project

I ran
`make menuconfig`
and just adapted the serial communication and the device's flash size to 4MB.

But running make (flash) I get the following error:
```
CC build/main/eq3_timer.o
C:/esp32_mqtt_eq3-master/main/eq3_timer.c:21:24: fatal error: controller.h: No such file or directory
compilation terminated.
make[1]: *** [/c/Users/toffe/Documents/esp-idf/make/component_wrapper.mk:286: eq3_timer.o] Fehler 1
```
In general, build instructions would be really nice 👍 
Thanks!

katanapod, 2018-10-07:
Same Problem here.

Tried the released BIN with my own compliled Partitions etc. 
It tries to start but run into a Boot loop.

 Are you active on this Repo and update it to last IDF ?

headcloudmonkey, 2018-10-28:
Experiencing the same issue when trying to build. Has anyone managed to get it working? Is the controller.h part of another project somewhere that we're missing?

ul-gh, 2019-01-02:
Hi,

I just now built the binary with the specific versions of esp-idf and esp-mqtt according to pripro's comments.

Regarding the necessary modifications, I, too, can confirm that the currently sold versions of the eQ-3 radiator thermostat use device name "CC-RT-BLE" without the "M".

Also, simply commenting out all code blocks for CONFIG_MQTT_SECURITY_ON did not work for me, the code still seemed to choose the SSL port, which I did not look further into. However, what works for me is just commenting out the #define CONFIG_MQTT_SECURITY_ON 1 in file esp-idf/components/esp-mqtt/include/mqtt_config.h, line 7. Of course this is only a temporary fix.

Thank you very much already, looking forward to a compatibility upgrade for the current release version of the ESP-IDF.

Regards,
Ulrich

MeisterQ, 2019-01-04:
Hello Ulrich,

can you send me your binary file?

ul-gh, 2019-01-05:
Binary release with added bluetooth device ID "CC-RT-BLE" for ESP32-WROOM32 module.

@softypit: i did some modifications to main/eq3_gap.c accordingly, please feel free to pull from https://github.com/ul-gh/esp32_mqtt_eq3.

MeisterQ, 2019-01-05:
Thank you Ulrich!

rmtucker, 2019-01-09:
> Binary release with added bluetooth device ID "CC-RT-BLE" for ESP32-WROOM32 module:
> https://github.com/ul-gh/esp32_mqtt_eq3/releases
> 
> @softypit: i did some modifications to main/eq3_gap.c accordingly, please feel free to pull from https://github.com/ul-gh/esp32_mqtt_eq3.

Trying your binaries but on reboot after firmware update i can not get it into AP mode and it just scans for eq3 even though i have not configured it yet.

ul-gh, 2019-01-10:
> i can not get it into AP mode

I can check later (on the weekend) if I messed up the compiled version. Did you erase the chip before programming?

@MeisterQ, the DNS for me worked out-of-the-box via DHCP; i supplied the DNS name for my MQTT broker and it was found immediately.

MeisterQ, 2019-01-10:
> > i can not get it into AP mode
> 
> I can check later (on the weekend) if I messed up the compiled version. Did you erase the chip before programming?
> 
> @MeisterQ, the DNS for me worked out-of-the-box via DHCP; i supplied the DNS name for my MQTT broker and it was found immediately.

@ul-gh 
The misstake was to enter :1883 after the ip of my broker.

Floyddotnet, 2019-02-24:
This issue is fixed in pull-request #14

closed 2019-09-20 by softypit