softypit/esp32_mqtt_eq3 #3: Flashing with esptool

hankman13, 2018-12-12:
Hi,
I just wanted to flash your binary file from the release to an ESP32, but I didn't get it right working.
Could you provide the right flashing command for the ESP32? I would like to do this the esptool.
Thank you!

MeisterQ, 2019-01-04:
Hi, do you have any more informations? Maybe a .bin File i could flash on my ESP?

hankman13, 2019-01-07:
I used the .bin File from the release page.
https://github.com/softypit/esp32_mqtt_eq3/releases/download/v1.0/eq3_trv_control.bin

softypit, 2019-01-12:
Hi Hankman,
Did you get this working? I uploaded the new binary with partition config and bootloader. I have never uploaded directly to the esp so I can't offer any advice there. I always build with the idf tools and upload using 'make flash' from the command line.
I believe you should be able to use the esp32 download tool with these files.

hankman13, 2019-01-14:
Hi softypit,
I tried following, I've downloaded the binaries and tried a command which I found in the [Ka-Radio32](https://github.com/karawin/Ka-Radio32) project.

`esptool.py \
   --chip esp32 --/dev/ttyUSB0 --baud 460800 --before default_reset \
   --after hard_reset write_flash -u --flash_mode dio \
   --flash_freq 40m --flash_size detect \
   0x1000 bootloader.bin \
   0x10000 eq3_trv_control.bin \
   0x8000 partitions.bin`

But I'm not sure if the address for the binaries is right (just copied and pasted it).
Anyway I can see something is running on the serial, and a AP is opened and I can connect.
I see the config page for wifi. 
More I will try later and report it.

hankman13, 2019-02-20:
So back again.
I set up the configuration, but afterwards I'm not sure if its working. Because I got no MQTT Message after all.
It would be great if the ESP32 could provide a status webpage with the information of used broker, Wifi SSID and topic name.
Will there be any information if I connect it to a serial monitor? (I did not tried out)

softypit, 2019-02-28:
Serial monitor will show connection information. This will be WiFi assoc indications, BLE scan results and mqtt connectivity attempts. There *should* be enough information there to debug any connectivity issues.

hankman13, 2019-02-28:
Thank you for that information, I will recheck my setup.

VHett, 2019-04-21:
On a Macbook and default python 2.7 I installed pyserial and esptool with pip and this worked for me:

sudo esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600  --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 bootloader.bin 0x10000 eq3_trv_control-3.bin 0x8000 partitions.bin

closed 2019-09-20 by softypit