softypit/esp32_mqtt_eq3 #4: No devices found: EQ-3 Bluetooth device name for MAC address discovery changed to "CC-RT-BLE"

ul-gh, 2019-01-02:
Hi, this change seems necessary for currently sold versions of the ELV/eQ-3, the device ID has changed or multiple IDs are available.

dwkit, 2019-02-17:
Hi, thanks for the good work on this Projekt.
I have flash your bin fork , my devices are CC-RT-BLE  and my esp32WROMM its no find any Devices.
In the main source code of Fork i found  "static const char remote_device_name[] = "CC-RT-M-BLE"; "
If this the problem? I would like to change it then. 
I can not compile the Fork the IDF had a lot of questions and at least it break up .
Its to many Questions how I can not answer.
Thanks for Help

dwkit, 2019-02-17:
eq3_main.c:32:24: fatal error: controller.h: No such file or directory
compilation terminated.

ul-gh, 2019-02-18:
Hi, the newer device ID is now included in these sources, this should work.
However, for the compilation issues, please, there is another thread about just this issue:
https://github.com/softypit/esp32_mqtt_eq3/issues/2
I guess there will be an update some time in the future, but until then, you basically need that exact older version IDF as in issue 2.

closed 2019-09-20 by softypit