softypit/esp32_mqtt_eq3 #5: Feature Request: RSSI (signal strength indication) for MAC addresses to blacklist/whitelist

ul-gh, 2019-01-03:
Hi, 

let me first thank you already for this code which is just what I was looking for to use the Equiva eQ-3 thermostats in conjunction with a soon-to-be roll-your-own centralised Node Red based control.

In my installation, the Bluetooth LE range of these thermostats is very poor and I need multiple of these ESP32 boards for full coverage.

In this situation with overlapping reach of the ESP32 MQTT EQ3 bridge modules, thermostats are discovered multiple times by the bridge modules. I am not sure if this is a battery drain issue for the thermostats, but anyways this would be a nice-to-have feature in order to choose the optimum one from different ESP32-MQTT-BLE bridge modules:

- Provide a signal strength indication for the discovered MAC addresses via MQTT or the web config frontend.

That way, the user can choose the ESP module providing the most reliable connection for any given installation location.

Regards, Ulrich

softypit, 2019-01-08:
Hi Ulrich,

I just had a look and it seems pretty simple to add rssi indication to the list of discovered devices. My preference would be to make the 'devlist' report in json i.e. {"devices":[{"rssi":-90,"bleaddr":"00:00:00:00:00:00"},{"rssi":-85,"bleaddr":"01:01:01:01:01:01"}]}. 
That should make it easier to process the responses from multiple ESPs and match up the reported devices to see which ESP has the best signal strength for each device. 
Let me know if you have a better suggestion.

Multiple ESP32s in range of a single trv should not cause a problem for battery life. The blescan is passive and simply looks for the beacons that each valve will be broadcasting at all times. 
I have 8 EQ3 BLE valves in my house with 2 ESP32s running esp32_mqtt_eq3 and a raspberry pi running eq3-mqtt-rpi. Some of the valves are visible from more than one eq3-mqtt bridge and I have to guess (trial and error) which bridge to use for best reliability. Currently I have had these running for over a year and not replaced any batteries yet.

Regards.

Paul.

softypit, 2019-01-12:
Hi Ulrich,
hopefully the code I just committed will be useful to you. Please let me know if it works as you expected. I have only run it on my development esp so far but will load it on to my valve controllers tomorrow.
Regards.
Paul.

ul-gh, 2019-01-13:
Thanks - that was quick!

I did a quick test and the RSSI indication via MQTT seems to be working
as intended. I tested with two TRVs, but will do some more testing next
week.

closed 2019-09-20 by softypit