softypit/esp32_mqtt_eq3 #10: OFF-Mode

yosh7777, 2019-01-15:
Hi, 
1st thanks for your great work. i'm currently using the thermostat with homebridge on two raspberry pis and planning to move to esp32, since i had some bluetooth connection issues. i can see, there's a off-mode in the homebridge-thermostat code by sending 0x41, 0x09 to the device. just wondering, why you didn't have it in your code. 

best regards,
Joachim

softypit, 2019-01-15:
Hi Joachim,
code 0x41, 0x09 is actually a settemp of 4.5 degrees although the valve will always operate at 5.0 degrees regardless of whether it is off to prevent frost. So I think the difference between 5.0 degrees and 4.5 degrees (off) is that the display shows 'OFF' but otherwise all operation is the same as for 5.0 degrees.
Currently the application rejects values below 5.0 and above 29.5 so cannot set 'OFF' or 'ON' mode.
Having said that it would be a trivial change to add off/on commands so I will try to do that for the next release.
Regards.
Paul.

softypit, 2019-03-03:
I just committed some changes which enable 'on' and 'off' mqtt commands. 'on' will fully open the valve and 'off' will close it (the actual set-temperature will be 5 degrees to prevent frost).

closed 2019-10-18 by softypit