softypit/esp32_mqtt_eq3 #12: Iget some Errors when i look at the serial connection

SigiLak, 2019-01-16:
my Esp-wroom32 get only one EQ3 Thermostat and shows some failure:

E (756) BT: btc_gattc_call_handler()

E (30766) BT: btc_search_callback  BLE observe complete. Num Resp 1

E (37916) BT: slave connection parameters update failed, the parameters are out of range

E (39156) BT: No pending command

E (68956) BT: btc_search_callback  BLE observe complete. Num Resp 1

E (74466) BT: L2CA_CancelBleConnectReq - no connection pending

E (74466) BT: p_conn is NULL in bta_gattc_conn_cback

E (74466) EQ3_MAIN: open failed, status 133

E (37916) BT: slave connection parameters update failed, the parameters are out of range

E (39156) BT: No pending command

E (41956) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016

E (68956) BT: btc_search_callback  BLE observe complete. Num Resp 1

E (74466) BT: L2CA_CancelBleConnectReq - no connection pending
E (74466) BT: p_conn is NULL in bta_gattc_conn_cback

E (74466) EQ3_MAIN: open failed, status 133

E (989876) BT: slave connection parameters update failed, the parameters are out of range

E (990896) BT: notification already registered

E (991106) BT: No pending command

E (996966) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016

E (1022296) BT: btc_search_callback  BLE observe complete. Num Resp 1

E (1029476) BT: L2CA_CancelBleConnectReq - no connection pending

E (1029476) BT: p_conn is NULL in bta_gattc_conn_cback

E (1029476) EQ3_MAIN: open failed, status 133

E (1119326) BT: slave connection parameters update failed, the parameters are out of range

E (1120336) BT: notification already registered

E (1120566) BT: No pending command

E (1124476) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016

Sigi

softypit, 2019-01-19:
I assume you are sending commands to a trv when some of these errors occur? The BLE observe complete are normal and appear after a scan. The others look like failed connection attempts. How close are the trvs to the ESP?

SigiLak, 2019-01-20:
Hi!
I think the Position was the Problem.
So i changed the esp to another Room an so i was able to catch more Thermostats!
great work
thx
Sigi

closed 2019-01-20 by SigiLak