softypit/esp32_mqtt_eq3 #25: Not responding anymore

MeisterQ, 2019-05-04:
Hello,

my TRVs worked very well last time, but since afew days its not working anymore.

I changed the batterieys of my valves and after startup of my esp its scanning for the valves, its publishing the list of found valves, but when i try to send any commands it shows following:

I (585803) mqtt: [APP] Publish topic: /Eq3radin/trv
I (585813) mqtt: Handle trv msg
I (585813) EQ3_MAIN: Requested address:
I (585823) EQ3_MAIN: 00 1a 22 0e ed 99
I (585823) EQ3_MAIN: Add queue end
I (585833) tmr: Start timer 1000 mS
I (588333) EQ3_MAIN: Timer0 event
I (588333) tmr: Start timer 1000 mS
I (590833) EQ3_MAIN: Timer0 event
I (590833) tmr: Start timer 1000 mS
I (593333) EQ3_MAIN: Timer0 event
I (593333) tmr: Start timer 1000 mS
I (595833) EQ3_MAIN: Timer0 event
I (595833) tmr: Start timer 1000 mS
I (598333) EQ3_MAIN: Timer0 event
I (598333) tmr: Start timer 1000 mS
I (600833) EQ3_MAIN: Timer0 event
I (600833) tmr: Start timer 1000 mS
I (603333) EQ3_MAIN: Timer0 event
I (603333) tmr: Start timer 1000 mS

Thats it.

softypit, 2019-05-04:
Which version of software are you running? 
There was an issue that floyddotnet fixed a while back where the command queueing could get stuck. It's possible you are seeing that issue. If this is the case newer software should fix it.

MeisterQ, 2019-05-05:
Hey,

i think im using this version https://github.com/softypit/esp32_mqtt_eq3/issues/13#issuecomment-466774419

Is it an old one?

softypit, 2019-05-07:
There are newer versions since this. Floyddotnet added further fixes and the latest version in this repo also contains his fixes. I would recommend upgrading. If you apply the latest binary update from this repo it adds ota update support so any future updates can be done with http over wifi. You will need to re-apply your configuration if you use this new version however.

MeisterQ, 2019-05-07:
I downloaded and flashed the new files. Well. Till now its working. Lets see.

closed 2019-09-20 by softypit