softypit/esp32_mqtt_eq3 #27: Port of MQTT

ivankhokhryakov, 2019-08-21:
Hi,
maybe my question is stupid, but I can't solve a problem.

I have Mosquito on my server with port 1883 and ip 192.168.2.22 

ESP32 has ip 192.168.2.33 and port 1883 as a deafult. I try to connect with command:
mosquitto_sub -h 192.168.2.33 -p 1883 -u myusername -P mypass -t "mytopicnameradout/devlist"

It doesn't work. I have a message:
Connection refused

What should I do?

Thank for your answers!

ivankhokhryakov, 2019-08-22:
Small update. I can't reach a config page too: 192.168.2.33/config
Message is the same: Connection refused

I have flashed with esptool:

esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 eq3_trv_control.bin 0x8000 partitions.bin

closed 2019-08-29 by ivankhokhryakov

maats, 2019-10-16:
Hey, I can see you already closed an issue but without any comment. Have you managed to resolve your problem? I have the same issue where I'm getting "Error: Connection refused" using console command and I can't reach my broker. Access from UI works okay but I'd like to know how to configure it for Home Assistant communication.

Could you tell more about your resolution, how did you solve it?

softypit, 2019-10-18:
Hi Maats,
it looks like the original issue was with ip broker vs client connectivity. The MQTT publish must go to the broker and the ESP is a client and unable to accept connections from other clients. Is this the same issue you are seeing?
If not please could you open a new issue explaining your configuration and problem.

Regards.

maats, 2019-10-19:
Hey @softypit,

I've managed to get this working and you're right with what you mentioned here. Initially I misunderstood behaviour of ESP32 with your application because I thought I need to attach RPi with Home Assistant as a client to ESP32 with your application as a broker so I set separate IP address in ESP32 MQTT settings and in RPi HA settings I set this MQTT IP which was completely wrong. Today I had a time so I read a documentation deeper and watched some YouTube videos about MQTT to understand it and after this I figure out that I need to set up locally Mosquitto in Home Assistant on RPi4 as a broker which I need to run and after running this service next I set in esp32_mqtt_eq3 settings in MQTT address IP of my RPi so then ESP32 got connected to Mosquitto broker correctly. I was able to set temperatures on my valves or to get the status after setting it. Later I also figure out from your documentation I can't invoke status without calling 'setter', it's a pity but as it's described I'll try to make workaround by calling setter e.g. with auto/manual state every 5 mins plus when publish message will go then I'd like to handle it as well. I'm not so familiar so far with Home Assistant automations but I hope that step-by-step will get this working as I need. Thanks guys for your effort in developing this solution.

Cheers,
maats

softypit, 2019-10-19:
I'm glad you got it working @maats. 

good luck with your home assistant journey. When you get your system working please feel free to add a wiki entry explaining how to do it for others.

Regards.
Paul.

maats, 2019-10-19:
Sure, will try to provide documentation once I get this working as I want. It may not be in the nearest time because unfortunately I'm a bit limited with my time and it's hard to find the time for configuring HA on RPi.