softypit/esp32_mqtt_eq3 #28: Compiling Issue /main/eq3_wifi

JBrumby, 2019-09-17:
Sorry I can't compile and I don't understand what to do. It seams a code problem, with sprint  %s 
Can anybody help.

Thanks in advance
Brumby
.
.
.
AR build/lwip/liblwip.a
CC build/main/eq3_main.o
CC build/main/eq3_wifi.o
In file included from /home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_wifi.c:28:
/home/peter/Arduinosrc/eq3esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
CC build/main/eq3_gap.o
CC build/main/eq3_bootwifi.o
In file included from /home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:16:
/home/peter/Arduinosrc/eq3esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'mongoose_serve_config_page':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:238:18: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             ibuf = inet_ntop(AF_INET, &connectionInfo.ipInfo.ip, ipbuf, sizeof(ipbuf));
                  ^
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:240:18: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             gbuf = inet_ntop(AF_INET, &connectionInfo.ipInfo.gw, gwbuf, sizeof(gwbuf));
                  ^
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:242:18: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             mbuf = inet_ntop(AF_INET, &connectionInfo.ipInfo.netmask, maskbuf, sizeof(maskbuf));
                  ^
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'bootWiFi':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:975:5: warning: 'esp_event_loop_init' is deprecated [-Wdeprecated-declarations]
     ESP_ERROR_CHECK(esp_event_loop_init(esp32_wifi_eventHandler, NULL));
     ^~~~~~~~~~~~~~~
In file included from /home/peter/Arduinosrc/eq3esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:15:
/home/peter/Arduinosrc/eq3esp/esp-idf/components/esp_event/include/esp_event_legacy.h:227:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
           ^~~~~~~~~~~~~~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'mongoose_serve_status':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:354:26: error: '</td></tr> <tr><td>MQTT user...' directive writing 38 bytes into a region of size between 26 and 281 [-Werror=format-overflow=]
         sprintf(htmlstr, connectedstatus, connectionInfo.mqtturl, connectionInfo.mqttuser, connectionInfo.mqttpass, connectionInfo.mqttid, connected == true ? "connected" : "not connected");
                          ^~~~~~~~~~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:354:9: note: 'sprintf' output between 330 and 746 bytes into a destination of size 430
         sprintf(htmlstr, connectedstatus, connectionInfo.mqtturl, connectionInfo.mqttuser, connectionInfo.mqttpass, connectionInfo.mqttid, connected == true ? "connected" : "not connected");
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'mongoose_serve_config_page':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:244:22: error: '" /></td></tr> <tr><td>Gatew...' directive writing 124 bytes into a region of size between 36 and 656 [-Werror=format-overflow=]
     sprintf(htmlstr, selectap, sptr, pptr, murlptr, muserptr, mpassptr, midptr, connectionInfo.ntpenabled != 0 ? "checked=\"checked\"" : "", connectionInfo.ntpserver, connectionInfo.ntptimezone, ibuf == NULL ? nullstr : ibuf, gbuf == NULL ? nullstr : gbuf, mbuf == NULL ? nullstr : mbuf);
                      ^~~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:244:5: note: 'sprintf' output 1872 or more bytes (assuming 2492) into a destination of size 1995
     sprintf(htmlstr, selectap, sptr, pptr, murlptr, muserptr, mpassptr, midptr, connectionInfo.ntpenabled != 0 ? "checked=\"checked\"" : "", connectionInfo.ntpserver, connectionInfo.ntptimezone, ibuf == NULL ? nullstr : ibuf, gbuf == NULL ? nullstr : gbuf, mbuf == NULL ? nullstr : mbuf);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'mongoose_event_handler':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:489:41: error: '%s' directive writing up to 14 bytes into a region of size between 5 and 38 [-Werror=format-overflow=]
                 sprintf(request, "%s %s %s", devstr, cmdstr, valstr);
                                         ^~                   ~~~~~~
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:489:17: note: 'sprintf' output between 3 and 50 bytes into a destination of size 40
                 sprintf(request, "%s %s %s", devstr, cmdstr, valstr);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/home/peter/Arduinosrc/eq3esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'eq3_bootwifi.o' failed
make[1]: *** [eq3_bootwifi.o] Error 1
/home/peter/Arduinosrc/eq3esp/esp-idf/make/project.mk:584: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

softypit, 2019-09-19:
Hi,
it looks like you've got format-overflow checking enabled in your compiler. This has uncovered my lazy coding with regard to string lengths in the web server code.
So your compiler is working out the maximum string length it is going to print according to all of the arrays that get included and if all are at maximum size they will overflow the output array.

e.g. /home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c: In function 'mongoose_serve_status':
/home/peter/Arduinosrc/eq3esp/esp32_mqtt_eq3/main/eq3_bootwifi.c:354:26: error: '</td></tr> <tr><td>MQTT user...' directive writing 38 bytes into a region of size between 26 and 281 [-Werror=format-overflow=]
         sprintf(htmlstr, connectedstatus, connectionInfo.mqtturl, connectionInfo.mqttuser, connectionInfo.mqttpass, connectionInfo.mqttid, connected == true ? "connected" : "not connected");
Indicates that if the mqtturl, mqttuser, mqttpass and mqttid are at maximum length (256 + 64 + 64 + 32 characters) the output buffer will be too small and an overflow will occur.

I haven't seen this reported before. What IDF/compiler are you using?

I've updated with a few minor changes and briefly tested on one of my devices which seems to work OK. As I don't see the warnings/errors myself I cannot test that it will fix your issue. Please give this a try and let me know how you get on.

Paul.

JBrumby, 2019-09-22:
Toolchain path: /home/peter/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
It is the newest compiler chain, that is online available. It seems it has the check enabled. 
I search how to disable it.
Thanks

Floyddotnet, 2020-02-05:
sorry for the late response but you use esp-idf version 4.* .. with this version the event_loop was redesigned and the build-system was changed

please use the esp-idf version 3.*