softypit/esp32_mqtt_eq3 #30: Flash address for OTA

MeisterQ, 2019-09-29:
Hello,

whats the address for the OTA upload? I cant do it working.

For control im using 0x10000
for bootloader 0x1000
and for partitions 0x8000

but for the OTA file?

softypit, 2019-10-01:
Hi,
If you check out partitions.csv that should explain the layout.

MeisterQ, 2019-10-24:
So, its 0x2000 ?

softypit, 2019-10-25:
I'm not sure I understand what you mean. The ota is applied via the web server interface and the image position alternates between ota_0 and ota_1.
What are you using the addresses for?

MeisterQ, 2019-10-26:
I figured out the address.

What i meant was the SPI-Address.

For eq3_trv_control.bin it was 0x10000, for bootloader.bin it was 0x1000, for partitions.bin 0x8000 and the 0x for ota_data_initial.bin was written nowhere.

But in the partitions.csv there was written 0xd000. And this worked. Thank you.