softypit/esp32_mqtt_eq3 #33: No connection to AP possible

wkraft-fablabka, 2019-10-18:
I compiled this project (VSCode/PlatformIO)
espressif32  @ 1.11.0 
tool-esptoolpy   @ 1.20600.0
toolchain-xtensa32 @ 2.50200.80 
framework-espidf  @ 3.30300.190916

It works fine, when the ESP32 is itself in AP mode. However I am not able to connect to any AP as station.  When I compile some of the espidf example programs, they connect immediately in station mode to all tested AP.

Are there others, having similar problems?

softypit, 2019-10-19:
Hi,
I've not used the combination of packages you listed above so cannot answer this directly.
Have you monitored the output from the serial console? This may show some useful debug information.

wkraft-fablabka, 2019-10-22:
Serial output is not really a help.
Before it logs the first time  ["Connecting to access point"](https://github.com/softypit/esp32_mqtt_eq3/blob/4820028387b3bcde5829ae05c95d6eba5b47e751/main/eq3_bootwifi.c#L853) 
which should by my understanding the station initialisation, I see already this lines:
```
D (14110) websrv: << bootWiFi2
D (14110) websrv: << bootWiFi
D (16160) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:TEST-SSID, ssid_len:11, bssid:00:00:00:00:00:00, reason:201,wifi reason: no ap found
V (16160) event: enter default callback
```
and after that there is a repetition of this log lines, until it changes to AP mode for 5 minutes:

```
V (26760) esp_adapter: thread sem get: sem=0x3ffdff80
V (26770) event: exit default callback
D (26770) websrv: Station disconnected started
D (26770) websrv: - Connecting to access point "TEST-SSID" ...
V (26780) esp_adapter: thread sem get: sem=0x3ffdff80
D (28840) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:TEST-SSID, ssid_len:11, bssid:00:00:00:00:00:00, reason:201,wifi reason: no ap found
V (28840) event: enter default callback
V (28840) tcpip_adapter: check: local, if=0 fn=0x401244f4
D (28850) tcpip_adapter: if0 start ip lost tmr: enter
D (28850) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffdfe8c interval=120 ip=0
V (28860) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (28870) tcpip_adapter: check: remote, if=0 fn=0x401244f4
```

softypit, 2019-10-23:
Looking at your trace the ssid_len is reporting as 11 and should be 9. It looks like I didn't initialise the sta_config structure properly which could be the issue. I don't have time to rebuild and test right now (and I don't see the issue so may not be able to reproduce) but you could try initialising the sta_config (line 867 of eq3_bootwifi.c) by adding the line indicated below:
        wifi_config_t sta_config;
+++  memset(&sta_config, 0, sizeof(wifi_config_t));
        sta_config.sta.bssid_set = 0;
        memcpy(sta_config.sta.ssid, pConnectionInfo->ssid, SSID_SIZE);
        memcpy(sta_config.sta.password, pConnectionInfo->password, PASSWORD_SIZE);

Please let me know if you try this and whether it works.

yosh7777, 2019-11-30:
your last beta fixes this problem. thanks!