softypit/esp32_mqtt_eq3 #34: Reboot

Lordcybertracker, 2019-11-02:
Hallo,
i have the Problem:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6392
load:0x40078000,len:11308
load:0x40080400,len:6672
entry 0x40080764
[0;32mI (29) boot: ESP-IDF v3.3-beta1-432-g6c5481977-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 22:36:21[0m
[0;32mI (39) boot: Enabling RNG early entropy source...[0m
[0;32mI (39) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (81) boot:  3 ota_0            OTA app          00 10 00010000 00177000[0m
[0;32mI (89) boot:  4 ota_1            OTA app          00 11 00190000 00177000[0m
[0;32mI (96) boot: End of partition table[0m
[0;32mI (100) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3b024 (241700) map[0m
[0;32mI (194) esp_image: segment 1: paddr=0x0004b04c vaddr=0x3ffbdb60 size=0x03334 ( 13108) load[0m
[0;32mI (199) esp_image: segment 2: paddr=0x0004e388 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (201) esp_image: segment 3: paddr=0x0004e790 vaddr=0x40080400 size=0x01880 (  6272) load[0m
[0;32mI (212) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xe6cd4 (945364) map[0m
[0;32mI (550) esp_image: segment 5: paddr=0x00136cf4 vaddr=0x40081c80 size=0x148b0 ( 84144) load[0m
[0;32mI (598) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (598) boot: Disabling RNG early entropy source...[0m
[0;32mI (598) cpu_start: Pro cpu up.[0m
[0;32mI (602) cpu_start: Application information:[0m
[0;32mI (607) cpu_start: Project name:     eq3_trv_control[0m
[0;32mI (612) cpu_start: App version:      1.20-28-gb312d25[0m
[0;32mI (618) cpu_start: Compile time:     22:01:31[0m
[0;32mI (623) cpu_start: Compile date:     Apr  5 2019[0m
[0;32mI (628) cpu_start: ESP-IDF:          v3.3-beta1-432-g6c5481977-dirty[0m
[0;32mI (635) cpu_start: Starting app cpu, entry point is 0x400810ec[0m
[0;32mI (628) cpu_start: App cpu up.[0m
[0;32mI (646) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (653) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (659) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (665) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (671) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (677) heap_init: At 3FFCE6E8 len 00011918 (70 KiB): DRAM[0m
[0;32mI (683) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (690) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (696) heap_init: At 40096530 len 00009AD0 (38 KiB): IRAM[0m
[0;32mI (702) cpu_start: Pro cpu start user code[0m
[0;32mI (50) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (113) BTDM_INIT: BT controller compile version [fe7ced0][0m
[0;32mI (113) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (213) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0[0m
[0;32mI (583) EQ3_MAIN: REG_EVT[0m
[0;32mI (583) tmr: Start timer 1000 mS[0m
[0;32mI (593) EQ3_CTRL: scan start success[0m
[0;32mI (913) EQ3_CTRL: Found device CC-RT-BLE - rssi -82, ble_addr_type: 0[0m
[0;32mI (913) EQ3_CTRL: 00 1a 22 0d d9 e6 [0m
[0;32mI (3083) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (3083) tmr: Start timer 1000 mS[0m
[0;32mI (5583) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (5583) tmr: Start timer 1000 mS[0m
[0;32mI (8083) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (8083) tmr: Start timer 1000 mS[0m
[0;32mI (10583) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (10583) tmr: Start timer 1000 mS[0m
[0;32mI (13083) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (13083) EQ3_MAIN: Init wifi[0m
I (13083) wifi: wifi driver task: 3ffdd9f8, prio:23, stack:3584, core=0
I (13083) wifi: wifi firmware version: 66a98c0
I (13093) wifi: config NVS flash: enabled
I (13093) wifi: config nano formating: disabled
[0;32mI (13093) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (13103) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (13143) wifi: Init dynamic tx buffer num: 32
I (13143) wifi: Init data frame dynamic rx buffer num: 32
I (13143) wifi: Init management frame dynamic rx buffer num: 32
I (13143) wifi: Init static rx buffer size: 1600
I (13153) wifi: Init static rx buffer num: 10
I (13153) wifi: Init dynamic rx buffer num: 32
[0;32mI (13153) websrv: Config version 200 found[0m
[0;32mI (13163) websrv: Network config present - becoming client[0m
I (13183) wifi: mode : sta (24:6f:28:16:7a:8c)
I (15933) wifi: new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (16923) wifi: state: init -> auth (b0)
I (16933) wifi: state: auth -> init (8a0)
I (16933) wifi: new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (22323) wifi: new:<11,2>, old:<11,0>, ap:<255,255>, sta:<11,2>, prof:1
I (22323) wifi: state: init -> auth (b0)
I (22323) wifi: state: auth -> assoc (0)
I (22333) wifi: state: assoc -> run (10)
I (22423) wifi: connected with FRITZ!Box 7530 MP, channel 11, bssid = 44:4e:6d:61:58:d1
I (22423) wifi: pm start, type: 1

[0;32mI (23083) event: sta ip: 192.168.178.84, mask: 255.255.255.0, gw: 192.168.178.1[0m
[0;32mI (23083) websrv: ********************************************[0m
[0;32mI (23083) websrv: * We are now connected and ready to do work![0m
[0;32mI (23093) websrv: * - Our IP address is: 192.168.178.84[0m
[0;32mI (23103) websrv: ********************************************[0m
[0;32mI (23103) EQ3_MAIN: WiFi network connected
[0m
[0;32mI (23113) EQ3_MAIN: SNTP not enabled
[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400014fd  PS      : 0x00060a30  A0      : 0x800e2888  A1      : 0x3ffdbdc0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffdc090  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a20  A13     : 0x3ffb65f0  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffdbdc0 0x400e2885:0x3ffdbdd0 0x400df879:0x3ffdc0e0 0x400d6317:0x3ffdc1a0 0x400d4c3b:0x3ffdc220 0x400d6a79:0x3ffdc2c0 0x401480b9:0x3ffdc300 0x401480fd:0x3ffdc320 0x400938b1:0x3ffdc370

Rebooting...

softypit, 2019-11-04:
This looks like it may be a problem with the configuration. Please check you have valid data for mqtt url and mqtt id in the configuration. You can force the application into AP mode by pulling GPIO0 low (depending on your esp32 board this will likely be a push-button) during boot. Then you can reconfigure the application.
Please let me know if this fixes the problem (and what values you were using for url/id) as I can add some checking in the code to prevent the continuous reboots if the configuration is bad.

softypit, 2019-11-10:
I've released a new beta (1.49-beta). I don't know if this will fix the issue but please give it a try and let me know.

Regards.
Paul.

jdsnape, 2020-03-27:
In case it helps anyone else, I had this issue when I forgot to set a value for MQTT ID. It might be useful if the web interface didn't let you press submit if you haven't filled out all the mandatory fields?