softypit/esp32_mqtt_eq3 #35: Unexpected entry into AP mode

smarthome-w, 2019-11-07:
There is problem with unexpected entry into AP mode. Logs:

```
[0;32mI (122964367) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (122964367) tmr: Start timer 1000 mS[0m
[0;32mI (122966867) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (122966867) EQ3_MAIN: Becoming WiFi client again
[0m
I (122966867) wifi: mode : sta (84:0d:8e:0e:9f:e0)
[0;31mE (122966877) MQTT_CLIENT: Error transport connect[0m
[0;32mI (122966877) mqtt: MQTT disconnected - wait for reconnect[0m
[0;32mI (122986877) mqtt: Other event id:7[0m
[0;31mE (122986877) MQTT_CLIENT: Error transport connect[0m
[0;32mI (122986877) mqtt: MQTT disconnected - wait for reconnect[0m
[0;32mI (123006877) mqtt: Other event id:7[0m
[0;31mE (123006877) MQTT_CLIENT: Error transport connect[0m
[0;32mI (123006877) mqtt: MQTT disconnected - wait for reconnect[0m
I (123022357) wifi: mode : softAP (84:0d:8e:0e:9f:e1)
I (123022357) wifi: Total power save buffer number: 16
I (123022357) wifi: Total power save buffer number: 16
[0;32mI (123022367) websrv: **********************************************[0m
[0;32mI (123022367) websrv: * We are now an access point and you can point[0m
[0;32mI (123022377) websrv: * your browser to http://192.168.4.1[0m
[0;32mI (123022377) websrv: **********************************************[0m
[0;32mI (123022387) EQ3_MAIN: WiFi connection failed - entering AP mode for 5 minutes
[0m
[0;32mI (123022397) tmr: Start timer 1000 mS[0m
[0;32mI (123022407) websrv: **********************************************[0m
[0;32mI (123022407) websrv: * We are now an access point and you can point[0m
[0;32mI (123022417) websrv: * your browser to http://192.168.4.1[0m
[0;32mI (123022417) websrv: **********************************************[0m
[0;32mI (123022427) EQ3_MAIN: WiFi connection failed - entering AP mode for 5 minutes
[0m
[0;32mI (123022437) EQ3_MAIN: setnextcmd when timer running![0m
[0;32mI (123024387) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (123024387) tmr: Start timer 1000 mS[0m
[0;32mI (123026877) mqtt: Other event id:7[0m
[0;32mI (123026887) EQ3_MAIN: Timer0 event (nextcmd.running=1, ble_operation_in_progress=0)[0m
[0;32mI (123026887) tmr: Start timer 1000 mS[0m
```
After reset esp32 is in normal (not AP) mode.

smarthome-w, 2019-11-07:
The very simple way to reproduce:

- connected to the router, working
- disable network (turn off router)
- esp32 is in AP mode after some time
- connect network (turn on router)
- network is available
- esp32 is in AP mode, no 5 minutes timeout (or longer) is applicable

After restart esp32 starts with normal settings (till next network unstability).

Software: build from latest commit

softypit, 2019-11-10:
I've released a new beta which may help here (v1.49-beta). I've not tested your issue specifically as I haven't had a chance yet.
I believe there was an issue where wifi mode wasn't cleared-down properly when switching from STA to AP and vice-versa. Could you try this version and let me know if it fixes things.

Regards.
Paul.

smarthome-w, 2019-11-11:
I've tested it and it looks good. esp32 was in AP mode and then restarted WIFI and connected in normal mode as expected.

I've prepared solution with hard reset, but your is much better.

My small suggestions for beta:

- change time for AP mode in case of problems from 5 minutes to 1 or 2
- increase logging in the loop: `ESP_LOGI(GATTC_TAG, "Timer0 event (nextcmd.running=%d, nextcmd.cmd=%d, nextcmd.countdown=%d, ble_operation_in_progress=%d)", nextcmd.running, nextcmd.cmd, nextcmd.countdown, ble_operation_in_progress);` (added cmd and countdown).