softypit/esp32_mqtt_eq3 #36: Add information on how to flash to ESP32

stefsolar, 2019-11-09:
Hey guys,

I'm trying to flash the code to my ESP since a while now, which is why I think it could be helpful to others to add this to the readme.
I started by installing Espressifs IDF and then trying to build the project, which didn't work ("CMakeLists.txt not found in project").

Next, I discovered the "Releases" and tried to flash the binaries using esptool.py, but then got stuck because I didn't know which ones to flash, and to which registers. I started with `esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 bootloader.bin` and am now a bit unsure on how to continue.

Anyways, I'd really appreciate some comments, also, happy to add to the readme once I know how "end-users" are meant to use this seemingly very nice software :-)

Best, Stef

softypit, 2019-11-10:
Hi Stef,
the command used by idf to flash the firmware is:
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 ota_data_initial.bin 0x1000 bootloader.bin 0x10000 eq3_trv_control.bin 0x8000 partitions.bin

Regards.
Paul

stefsolar, 2019-11-10:
Awesome, that worked! Thank you.

softypit, 2019-11-10:
Thanks Stef,
I've pulled your readme change and released a new beta with a few changes (v1.49-beta). This has the most recent fixes to a few issues.

Regards.
Paul.

stefsolar, 2019-11-14:
Thanks dude :-)

closed 2019-11-14 by stefsolar