softypit/esp32_mqtt_eq3 #38: Version 1.2 memory addresses

pmknowles, 2019-11-14:
I'm having some issues with v1.49. NTP gives dates in 1970 and scans saying devices found but nothing in the list.
I want to downgrade to the release version but the memory addresses in the readme are for the beta version (OTA for example). Using OTA to downgrade to 1.2 doesn't seem to work.
I also have one device which works fine connected to the PC but won't work from a USB PSU which is weird.
Great bit of software though. Managed to control radiators from openHAB which is what I wanted.

softypit, 2019-11-14:
The command used by idf for flashing the 1.20 code is:
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 eq3_trv_control.bin 0x8000 partitions.bin.
Hope this helps.

Do you have any traces of the problems with v1.49? I'd like to make it a release at some point but if there are bugs it would be nice to squash them all first.

Paul.

pmknowles, 2019-11-19:
I'm not sure whether it's 1.49. My other ESP32 works with NTP and finds the correct time but the first doesn't.
What I though was a bug was actually with the ESP32 needing to be within about 2m to find the EQ3-BLE. When I was setting it up it found the radiator in the bedroom above but would not see it from within the room 4m away. I know that it's possible to change the power from the default but not sure where you set it in your code.
I have 3 radiators in WC, Utility Room and Kitchen. The ESP32 is in the Utility Room but only sees the one in the Utility Room but not in the other rooms about 2.5m away.

softypit, 2019-11-19:
Which ESP32 modules are you using?
For the NTP issue:
Do both connect to the access point successfully? are you using DHCP or fixed IP addresses? ESP needs a connection to an ntp server so it's worth checking it has internet access once it is associated with the AP.
For signal:
Are you able to see the EQ3 valves using a smartphone or computer at the same location as the ESP? 
It should be possible to adjust the transmit power of the ESP but I suspect the problem you have may be with receive sensitivity.

pmknowles, 2019-11-20:
I'm using 2 development modules from eBay (similar to the NodeMCU ESP8266 modules).
NTP - changing from static to DHCP solved this. I then put the static address in the same as the DHCP and it maintained the NTP. Not sure why this was an issue. The devices appeared in the router list so they should've had internet access.
If I use the Calor BT app I can see both upstairs radiators from Bedroom 1 and the 3 downstairs radiators and the Bedroom 2 radiator from the Utility Room

softypit, 2019-11-20:
Is it possible the gateway address was set wrong statically when you first tried? This would enable the ESP to work on WiFi but not access the timeserver.
Do you see the range problem with both ESPs? What is the reported rssi for the EQ3 that can be seen? I find the range of EQ3 units poor (which is typical for BLE) but no worse for ESP32 than calorBT on my phone.

pmknowles, 2019-11-21:
The static data was correct but I can now put it into static and it works. Possibly needs to use DHCP on first connection?
The rssi is 78% which isn't that bad but I have a nut on my keyring as presence detection which my Pi sees 10+m away through a 45cm stone wall. 
I'm going to try a different ESP32 to see if it's that.

closed 2019-11-21 by pmknowles

pmknowles, 2019-12-07:
Don't want to create a new issue as it may only be me. I am unable to set static IP addresses for these at all.
Even if I set the same address as DHCP I get connection refused. Do you know if anyone else is having this issue?
Also, when valve status is 100% open does that mean that the valve is or the actuator is? Setting the settemp to 5.0 gives me a 100% open when I would expect 0% (or 100% closed) but the actuator should be 100% extended.

softypit, 2019-12-07:
I've not seen any other reports of issues with static addresses. I just tried my spare ESP32 and had no problems. Do you have any more details? Does the ESP connect to the AP and the mqtt broker? 

The valve status percent indicates how far open the eq3 thinks the valve is. This is relative to the calibration that takes place during installation so 100% open means the pin is fully retracted in the EQ3 and 0% means it is driven down to its calibrated stop 'off' point. 
This value is always the 'current' state of the drive so if it is open and you change the required temperature to 5.0 you will see a report of 100% open. What should then happen is that the valve begins to close. If you poll it after a short while (e.g. with the manual/auto mode command) it should then show as being closed (0%).