softypit/esp32_mqtt_eq3 #39: how many thermostats can be controlled

jensenbln, 2019-11-18:
sorry for the questions but i could not figure it out of the description:
is it possible to control more than one thermostat with this?

bye

Floyddotnet, 2019-11-18:
due to the ESP32's limited memory size, the exact number depends on many factors and is difficult to calculate accurately. many of them consume a few bytes more or less depending on the operating mode and configuration.

roughly estimate, you should not reach a limit with 50 thermostats per ESP yet.

softypit, 2019-11-26:
For valve control esp32_mqtt_eq3 does not store any context data persistently. Each command submitted has the address of the valve to control and the command data. In theory there is no limit to the number of EQ3 valves a single ESP32 can control although obviously they must all be within BLE range.
There is also no coded limit to the number of commands that esp32_mqtt_eq3 will attempt to add to the command queue. In reality this will be limited by available heap space although in looking into this I notice there is no checking of the malloc struct for new commands so the ESP will likely crash if too many outstanding commands are queued. I may add a limit in future and will certainly add a malloc-check to the next release.

electracks, 2020-03-26:
The integration of multiple EQ3 thermostats is only partly working. The problem is, that the answer from the thermostat is used to update all thermostats, because HA can not determine which thermostat sends the answer. I think you need to filter the MAC with a rule and trigger specific MQTT commands. Is it possible to add examples for adding multiple thermostats to HA?