softypit/esp32_mqtt_eq3 #40: Wiki article for integration to Home Assistant

jan-hendrikscholz, 2019-11-26:
Hi @softypit ,
thank you for your awesome project!!

I create a documentation for adding esp32_mqtt_eq3 to home assistant.
If you want, you can put it in your wiki as I cannot create a PR for it. Seems, that it's a common issue with github....

Here my article:
# Home Assistant configuration

Integration into Home assistant is possible using the natively provided MQTT HVAC integration of HA. As the MQTT topic for the states is different than the inputs for temperature and mode, automations are needed to re-write the MQTT messages.
## General structure:
 1. MQTT HVAC configuration
 2. Helper automation for rewriting MQTT commands for temperature control
 3. Helper automation for rewriting MQTT commands for mode selection

### MQTT HVAC configuration
Adjust the name, MQTT topics and MAC-address to your esp32_mqtt_eq3 configuration. 

```
   - platform: mqtt
     name: Badezimmer
    device:
     identifiers: '00:1A:22:0B:42:94'
    modes:
      - "off"
      - "auto"
      - "heat"
    precision: 0.5
    min_temp: 4.5
    max_temp: 29.5
    temp_step: 0.5
    mode_state_topic: "/badezimmerradout/status"
    mode_state_template: >
      {% if value_json.mode == "manual" %}
        {% if value_json.temp == "4.5" %}
          off
        {% else %}
          heat
        {% endif %}
      {% elif value_json.mode == "auto" %}
        auto
      {% endif %}
    mode_command_topic: "/ha_badezimmerradin/trv_mode"
    temperature_state_topic: "/badezimmerradout/status"
    temperature_state_template: "{{ value_json.temp }}"
    temperature_command_topic: "/ha_badezimmerradin/trv_temp"
    current_temperature_topic: "/badezimmerradout/status"
    current_temperature_template: "{{ value_json.temp }}"
    away_mode_state_topic: "/badezimmerradout/status"
    away_mode_state_template: >
      {% if value_json.mode == "holiday" %}
        on
      {% else %}
        off
      {% endif %}
      
```

### Temperature Helper
Here as well, adjust the MQTT topics and BT names to your configuration. Add it to the automations.yaml

```
- id: '1573053873261'
  alias: Badezimmer_Temperature_Helper
  description: ''
  trigger:
  - platform: mqtt
    topic: /ha_badezimmerradin/trv_temp
  condition: []
  action:
    service: mqtt.publish
    data_template:
      payload: "{% if trigger.payload == \"4.5\" %}\n  00:1A:22:0B:42:94 off\n{% else\
        \ %}\n  00:1A:22:0B:42:94 settemp {{trigger.payload}}\n{% endif %}"
      topic: /badezimmerradin/trv
```

### Mode Helper
Adjust the automation to fit your environment and append it to your automations.yaml.

```
- id: '1573063154330'
  alias: Badezimmer_Mode_Helper
  description: ''
  trigger:
  - platform: mqtt
    topic: /ha_badezimmerradin/trv_mode
  condition: []
  action:
    service: mqtt.publish
    data_template:
      payload: "{% if trigger.payload == \"heat\" %}\n  00:1A:22:0B:42:94 manual\n\
        {% else %}\n  00:1A:22:0B:42:94 {{trigger.payload}}\n{% endif %}"
      topic: /badezimmerradin/trv
```

softypit, 2019-11-26:
Thanks @jan-hendrikscholz, I'm glad you like it.

Many thanks for the article. I'm not familiar with HA so this is really useful for anyone wanting to integrate their EQ3 valves.
I've added you content to the wiki.

Regards.

jan-hendrikscholz, 2019-12-01:
Thank you @softypit for adding the article to your wiki. I'll make you aware, if some future adjustments are needed :-)
In case someone needs advice regarding this article, everybody may contact me :-)

closed 2019-12-01 by jan-hendrikscholz

lukasdietz, 2021-03-29:
When trying to add this to my configuration.yml I have encountered an error:
`Property identifiers is not allowed` .
Could you help me out resolving this issue?

jensenbln, 2021-11-22:
if i manually send: 
`mosquitto_pub -d -t "/heizungradin/trv_temp" -m "00:1A:22:06:XX:XX settemp 21.0"`
it works.
but if i use the above HA integration for MQTT-HVAC nothing happens.
the mosquitto command sent from HA is:
`/ha_heizungradin/trv_temp 21.0`
nothing happens.
_

maunsen, 2022-01-05:
If you want to add multiple Thermostats to one ESP you have to edit the MQTT HVAC configuration. 
Otherwise an update on temperature or mode on one thermostat will be done shown on all Thermostats.

`   - platform: mqtt
     name: Badezimmer
    device:
     identifiers: '00:1A:22:0B:42:94'
    modes:
      - "off"
      - "auto"
      - "heat"
    precision: 0.5
    min_temp: 4.5
    max_temp: 29.5
    temp_step: 0.5
    mode_state_topic: "/badezimmerradout/status"
    mode_state_template: >
      {% if value_json.mode == "manual" %}
        {% if value_json.temp == "4.5" %}
          off
        {% else %}
          heat
        {% endif %}
      {% elif value_json.mode == "auto" %}
        auto
      {% endif %}
    mode_command_topic: "/ha_badezimmerradin/trv_mode"
    temperature_state_topic: "/badezimmerradout/status"
    **temperature_state_template: >
      {% if value_json.trv == "00:1A:22:0B:42:94"  %}
        {{ value_json.temp }}
      {% endif %}**
    temperature_command_topic: "/ha_badezimmerradin/trv_temp"
    current_temperature_topic: "/badezimmerradout/status"
    **current_temperature_template: >
      {% if value_json.trv == "00:1A:22:0B:42:94"  %}
        {{ value_json.temp }}
      {% endif %}**
    away_mode_state_topic: "/badezimmerradout/status"
    away_mode_state_template: >
      {% if value_json.mode == "holiday" %}
        on
      {% else %}
        off
      {% endif %}
      `
You only want to change the mode and the temperature of the thermostat, that was changed.