softypit/esp32_mqtt_eq3 #41: Increase BLE range

polskafan, 2019-12-13:
Hi,

I want to control my EQ-3 valves (model 141771A1A) via MQTT, so I bought some Lolin32 ESP32 Lite Development boards and flashed them with your software.

However, I am having severe connection issues. I sometimes can't even connect to a valve that is just 1 metre away. I was hoping to cover my flat with two ESP32 boards (5 valves in 5 rooms, ~60m²).

I already tried to increase the BLE TX power as described here:
https://github.com/nkolban/esp32-snippets/issues/674

I am not 100% sure, where to set the power, but a subsequent read back reports all three power levels at 7, which should be the highest. It didn't really change anything for the BLE range  #though.

This are the two places, that I have tried to change BLE power:
```diff --git a/main/eq3_main.c b/main/eq3_main.c
index e1dd433..3835da8 100644
--- a/main/eq3_main.c
+++ b/main/eq3_main.c
@@ -1158,6 +1158,20 @@ void app_main(){
         ret = nvs_flash_init();
     }
     ESP_ERROR_CHECK( ret );
+
+    /* increase transmission power */
+    ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);
+    if (ret) {
+        ESP_LOGE(GATTC_TAG, "%s ble set power failed, error code = %x\n", __func__, ret);
+    }
+
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_P9);
+
+    ESP_LOGI(GATTC_TAG, "ble power levels = (default %d, adv %d, scan %d)",
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_DEFAULT),
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV),
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_SCAN));

     esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
     ret = esp_bt_controller_init(&bt_cfg);
@@ -1184,6 +1198,20 @@ void app_main(){
         return;
     }

+    /* increase transmission power */
+    ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);
+    if (ret) {
+        ESP_LOGE(GATTC_TAG, "%s ble set power failed, error code = %x\n", __func__, ret);
+    }
+
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_P9);
+
+    ESP_LOGI(GATTC_TAG, "ble power levels = (default %d, adv %d, scan %d)",
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_DEFAULT),
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV),
+        esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_SCAN));
+
     //register the callback function to the gattc module
     ret = esp_ble_gattc_register_callback(esp_gattc_cb);
     if(ret){
```

Are there any other things, I could try to get a more stable connection? Do other ESP32 boards have better BLE coverage or should I switch to a Raspberry Pi Zero as a heating controller?

softypit, 2019-12-13:
As far as I can tell range issues with these valves are primarily caused by the receive sensitivity of the ESP32 and transmit power of the valve itself. i.e. changing the transmit power of the ESP32 doesn't seem to make much difference.
What values of rssi do you see when the ESP32 scans for devices?
Are you getting good range using the calorBT app?
Also what errors do you see from the ESP32 serial output when valves are uncontactable?

polskafan, 2019-12-13:
This is the log of a valve working at around 1m distance and failing at 2m: https://pastebin.com/dnPyiNa9

The Calor app gives much better range, I can at least control the valves from the adjacent rooms from my phone.

softypit, 2019-12-14:
The rssi value seems OK - was this at 1m?
I see issues with connection occasionally and that was the reason for adding the retries. I suspect this is due to interference as I have a lot of 2.4GHz rf equipment around. I do get ranges of up to 8m though.
I'm using ESP32WROOM-based devices which have the ESP32 inside an EMC shield whereas the lite looks to have everything on a single board with no shielding. I don't know whether this will have any effect on signal integrity/range.
If you run the scan at 2m/3m etc does the ESP pick up the valves and show a reasonable rssi?

polskafan, 2019-12-17:
Yeah, the RSSI was at 1m. At 2m I can't pick up any of the valves with the Lolin32.

I just did a test with a Raspberry Pi Zero W in the center of my flat and can control all of the 5 valves at ranges up to 10m. The remaining dropouts seem to be related to radio interference. For example, I can't reach the valve furthest away when I use my Wifi.

I guess, I managed to buy the worst ESP32 boards for BLE when looking at the performance of the Calor App and the Raspberry Pi Zero W.

I'll try a WROOM-32 based board and see how it will perform.

polskafan, 2019-12-18:
Okay, I managed to source a WROOM-32 Module for testing (AZDelivery ESP32 NodeMCU). From my initial tests, it performs waaaay better than the Lolin32 Boards. I can now read the valves from up to 5-6m away with RSSI values of -85 to -90. Strategically placed, I can now control 2 modules at once.

However, the Raspberry Pi Zero W still outperforms the range of the new module and can control all of the 5 valves from one location.

Again, it doesn't seem to matter much, if I use the official build or my modified one with the increased BLE power.

polskafan, 2020-01-16:
I watched Andreas Spiess's youtube video on long range ESP32 connections. From there I figured, I'll give it a last try with a ESP32 module, that has an external antenna connector. I ordered the ESP32-DevKitC V4 from AliExpress and grabbed an external wifi antenna and a U.FL to IPX pigtail adapter.

As a result, I can now receive all of my valves from the center of my flat with reasonably good RSSI values:
`Device:	00:1A:22:12:xx:xx	rssi	-86
Device:	00:1A:22:12:xx:xx	rssi	-94
Device:	00:1A:22:12:xx:xx	rssi	-71
Device:	00:1A:22:12:xx:xx	rssi	-81
Device:	00:1A:22:12:xx:xx	rssi	-88
`

So, I guess this issue can be closed. It was completely hardware related.

softypit, 2020-01-17:
That's great news. 
If you have more details/pics/advice that would help others it would be great to have a wiki page about how to maximise coverage from a single ESP32 node.

yunnanpl, 2020-02-23:
Hi, I have 4 devices. 3x AZDelivery ESP32-WROOM-32 and 1x MakerHawk ESP32S with camera.
The signal power is similar for all, whereas the ESP32S has an external antenna connector (but nothing connected as signal is good enough).

Device:	00:1A:22	rssi	-82
Device:	00:1A:22	rssi	-77
Device:	00:1A:22	rssi	-88

Device:	00:1A:22	rssi	-80
Device:	00:1A:22	rssi	-91
Device:	00:1A:22	rssi	-89

Device:	00:1A:22	rssi	-90
Device:	00:1A:22	rssi	-79
Device:	00:1A:22	rssi	-78

Device:	00:1A:22	rssi	-83
Device:	00:1A:22	rssi	-99
Device:	00:1A:22	rssi	-90
Device:	00:1A:22	rssi	-90
Device:	00:1A:22	rssi	-94

Still, I do not communicate with devices with rssi above 90 (usually :)), as this is not very stable. Distances between esp32 and eq3 differ, but there is usually a ~5m distance a wall between.
(Keep in mind that I have only 9x EQ3, but some ESP32 see them twice).

Saiuday9787, 2023-11-13:
Hiii, I am trying for LONG RANGE BLE (>300meters) using ESP32 C3 mini-1.

I tried esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P21);  at server side and client side, still i am able to get Line of site upto 100meters and able to tx via one floor concrete slab. still my requirement is not satisfied, 
Can you suggest on how to achieve above 300meters range.