softypit/esp32_mqtt_eq3 #42: Change hostname espressif

kroon040, 2019-12-17:
Hi,

Connected to my home wifi network, de esp32 is displayed as hostname espressif. It should be nice that it could be change in the webinterface or it should be called eq3-heatcontroller for example

regards Richard

softypit, 2019-12-17:
Hi Richard,

that's a good idea. I hadn't spotted this as I use static host entries in my dhcp server for my ESP32s. It shouldn't be too tricky so I'll add it for the next beta. 

If you've built from source you could try adding the following inside becomeStation() in eq3_bootwifi.c:
tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, "EQ3-heatcontroller");

I haven't tried it myself yet so cannot guarantee it will work.

Regards.
Paul.

kroon040, 2019-12-17:
HI,

I didn't build it,  I used the bins. So hopefully soon a new beta;-)

MarkStephan89, 2021-01-14:
hi, 
can you add a form into the config web page?

thank you :)