softypit/esp32_mqtt_eq3 #44: Set offset-Temp does'n work

xaziwb, 2020-01-08:
Hi,
I have a problem settig the offset temperature. The offset temperature stays unchanged and the temp is change to 4.5 °C.
Several tries - always the same.

see the Log file: 
[eq3.log](https://github.com/softypit/esp32_mqtt_eq3/files/4035471/eq3.log)

Regards
  Xander
Btw: Setting the offset temp is missing in the Web-Frontent.

softypit, 2020-01-11:
Hi Xander,
If you set the offset on the valve itself using the panel are the changes reflected when you poll with the ESP32? 
i.e. can you set the offset using the buttons on the valve then try using the mqtt command? Does the reported value in mqtt report the value you set from the panel or is it still 0.0?

Regards.
Paul.

xaziwb, 2020-01-12:
Hello Paul,

Thanks you for your reply.

I set the offset to 1.5°C with the button of the valve and it is reported correctly via MQTT:

2020.01.11 - 09:55:10 /x01radout/status {"trv":"00:1A:22:14:33:CE","temp":"21.0","offsetTemp":"1.5","valve":"0% open","mode":"auto","boost":"inactive","window":"closed","state":"unlocked","battery":"G
OOD"}

Then I have tried to change to 1.0°C with a MQTT-Client:
2020.01.11 - 10:24:12 /x01radin/trv 00:1A:22:14:33:CE offset 1.0

Answer:
2020.01.11 - 10:24:16 /x01radout/status {"trv":"00:1A:22:14:33:CE","temp":"4.5","offsetTemp":"1.5","valve":"0% open","mode":"auto","boost":"inactive","window":"closed","state":"unlocked","battery":"GO
OD"}

Regards
  Xander

softypit, 2020-01-12:
So it looks like the status is returned OK but the offset isn't setting properly. Do the other settings work? i.e. if you change the settemp/lock/unlock via mqtt  does that work correctly?

xaziwb, 2020-01-13:
Yes. If I try to change the offset it is unchanged but the temp is set to 4.5 °C.

In my OpenHAB Home Automation I use "unlock" the request the current settings. Manual sending "lock" works to. "lock" / "unlock" is therefore "lost" for my Application - but this is another problem.

Other Settings:
boost/unboost: OK
auto/manual: OK
on: OK (found it, but please add in the documentation)
off:  OK (also not in the docs. Or have I miss it?)
settime: not used until now. I want to uses NTP but it doesn't work. What is the correct timezone string for Germany?
scan: OK, but only if send a a payload with at least on character e.g. "/.../scan x". Without there is no answer. Don't ask how many tries I needed ...

softypit, 2020-01-14:
I found the bug in the software for the offset and have made a quick fix to test.
Are you using binaries or building from source?

xaziwb, 2020-01-15:
I use the binary 1.49-beta

softypit, 2020-01-17:
Sorry for the delay. I managed to wreck my build and had to revert and re-apply my changes.
I pushed an update with a fix for the offset command. It also includes offset support on the web gui. I also added 'on' and 'off' to the readme.
I dropped the binary build (1.51-beta). You should be able to apply the .bin directly from the web gui in the update software page.
Paul.

xaziwb, 2020-01-18:
Hey Paul,
Many thanks for Your work!
But at the moment i'm for skiing in the mountains.

Greetings from the “Dolomiten“
Xander

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: paul
Betreff: Re: [softypit/esp32_mqtt_eq3] Set offset-Temp does'n work (#44)
Datum: 17.01.2020, 22:44 Uhr
An: softypit/esp32_mqtt_eq3
Cc: Xander, Author

Sorry for the delay. I managed to wreck my build and had to revert and 
re-apply my changes.
I pushed an update with a fix for the offset command. It also includes 
offset support on the web gui. I also added 'on' and 'off' to the readme.
I dropped the binary build (1.51-beta). You should be able to apply the 
.bin directly from the web gui in the update software page.
Paul.

softypit, 2020-01-20:
Enjoy the cold :)

I think for your timezone you can use 'CET-1CEST,M3.5.0,M10.5.0/3'.

xaziwb, 2020-01-22:
Thanks

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: paul
Betreff: Re: [softypit/esp32_mqtt_eq3] Set offset-Temp does'n work (#44)
Datum: 20.01.2020, 21:48 Uhr
An: softypit/esp32_mqtt_eq3
Cc: Xander, Author

Enjoy the cold :)

I think for your timezone you can use 'CET-1CEST,M3.5.0,M10.5.0/3'.

xaziwb, 2020-01-26:
Hello Paul,
I'm back from holiyday. I have flashed the new version.
Setting the offset temp is better now. All values results in a correct setting. But in the reporting of the new value is an error :
The positive values from 0 to 3.5 are working correctly.
Setting to -0.5 gives -1.5; to -1 is OK; to -1.5 gives -2.5; to -2 is OK; to -2.5 gives -3.5; to -3 is OK; to -3.5 gives -4.5
The valve itself is allways set to the right temp. Only the reporting has an error at x.5 values.

Timezone:
At boot time the message "EQ3_MAIN: Waiting for system time to be set..." counts to 30 and then the ESP gives up. It seems there is no connection to a time server. I have testet with "pool.ntp.org" and "de.pool.ntp.org". The network parameters are set correctly.
Then I tested with the IP-address and the time is set. It seams to be a DNS problem. Any advice ...

Btw: Error messages / Warning:
I have noticed every "trv" command give 2 error messages and a warning. See Log file.
[eq3_20200126.log](https://github.com/softypit/esp32_mqtt_eq3/files/4113544/eq3_20200126.log)

softypit, 2020-02-09:
Hi Xander,
sorry for the delay - busy times at the moment.
I think I've figured out the offset value reporting. I can't easily debug it here as none of my valves actually report the offset. I assume I'm on older eq-3 valve software.
I'll try to post an updated software binary soon.

For the timezone issue - what wireless access point are you using? Is it a specific AP for smart-home devices or are you using a normal AP as used for other devices with internet connectivity? There are a few possibilities for DNS problems. The first to check is that the ESP is receiving a DNS server address in its DHCP offer. Also it's worth checking if any firewall rules prevent access to ip port 53. If you are able to use wireshark this should help here.

The error messages are something I see too. I've always intended to investigate them but not had the chance. I don't think the disconnect should be a warning as it occurs after an explicit disconnect request so it could just be some lingering trace within the IDF. The others don't seem to cause any operational problems so it should be OK to ignore them for now.

Paul.

Floyddotnet, 2020-02-09:
i open a pull request to fix the offset calculation #47

xaziwb, 2020-02-10:
Hi Paul,

Offset reporting:
I see Floyddotnet is working on it. Thanks!

Access Point:
It's a FritzBox 7580. On other clients connected via static IP to the AP DNS is working. DNS server is the IP of the AP.
But this is not a big issue to me since the "hard coded" time server is working.

Error messages:
All is working. It was only an info to You.

Floyddotnet, 2020-02-10:
This morning I had another idea that solves the whole problem artimetically and does the calculation without jumps completely #48 

but this is harder to read ... please Paul decide which version you want to adopt

softypit, 2020-03-03:
Hi, 
I merged the arithmetic offset value fix into the 1.55 beta pushed yesterday.
Let me know if you have any more issues with the offset setting/reporting.

Thanks @Floyddotnet for the fix.

xaziwb, 2020-03-06:
Hello,
I have flashed and tested the v1.56 beta. No issues found.
Many thanks!

PR #47 (2020-02-29): fix offset value calculation
this patch reimplements the offset calculation to get it right. Fixes #44

maybe there is a better way, but it works

negative values must be handled other than positive and -0.5 is a special case.

i test it with all possible values

    00 = -3.5
    01 = -3.0
    02 = -2.5
    03 = -2.0
    04 = -1.5
    05 = -1.0
    06 = -0.5
    07 =  0.0
    08 =  0.5
    09 =  1.0
    0A =  1.5
    0B =  2.0
    0C =  2.5
    0D =  3.0
    0E =  3.5
PR #48 (2020-02-29): fix offset value calculation (arithmetic version)
this patch reimplements the offset calculation to get it right. Fixes #44

i found arithmetic version without any jumps but this is a little bit harder to read in compare to #47 

please Paul decide which version you want to adopt

i test it again with all possible values with my local c compiler at work (without the xtensa tool chain)

    00 = -3.5
    01 = -3.0
    02 = -2.5
    03 = -2.0
    04 = -1.5
    05 = -1.0
    06 = -0.5
    07 =  0.0
    08 =  0.5
    09 =  1.0
    0A =  1.5
    0B =  2.0
    0C =  2.5
    0D =  3.0
    0E =  3.5
closed 2020-03-06 by xaziwb