softypit/esp32_mqtt_eq3 #45: DHT22 implementation

tkris, 2020-01-21:
Hi,

thank you for all your hard work with this application. Currently I´m using two Esp32 per room in my house.
One is running your software and the second one is running Micropython with a DHT22 temp/humidity sensor on the board. 
With Micropython it wasn´t too hard to get the DHT22 to work, but I´m a beginner and have no idea how to extend/modify your sourcecode.

Would it be possible for you to extend your application to support  the DHT22 sensor and publish the readings on two MQTT topics (for example .. /sensors/temp and /sensors/hum). If you don´t have an DHT22 for testing i would even provide for that.

Greetings
Kris

Floyddotnet, 2020-01-21:
hi tkris, 

buddebej already implement this: take a look at this commit: [buddebej@b645fb4](https://github.com/buddebej/esp32_mqtt_eq3/commit/b645fb4861fbdd2497e7c28fdd695d08a360776a) (Add support for dht22)

if you are a bit familiar with programming in general, you should find it easy to apply these changes.

apart from this, i recently start a new downstream project (based of this project) this supports xiaomi mijia temperature and humidity sensor, DHT22 sensors, and act as a 433 MHz gateway for RF temperature and humidity sensor, RF smoke detectors and RF window open detectors.

this project is currently in a early state and i this i can release a first version between Q3 and Q4.

tkris, 2020-01-21:
> hi tkris,
>
> buddebej already implement this: take a look at this commit: [buddebej@b645fb4](https://github.com/buddebej/esp32_mqtt_eq3/commit/b645fb4861fbdd2497e7c28fdd695d08a360776a) (Add support [...]

tkris, 2020-01-21:
Thank you for your reply,

Good to know that someone already implemented the support for the dht22.
Unfortunately i even don´t know how to work with the source-code an how to compile it.
I used the compiled *.bin Files to get the software up and running.

btw your project sounds very promising.  Would be great to try it when it´s ready.