softypit/esp32_mqtt_eq3 #52: No Configuration -WiFi-AP visible

Wetterprophet, 2020-03-02:
Hey there, thanks for all your hard work!
I really want to get this project running in my home :)

I got the esp32-image transferred with esptool, but with none of my devices, I can find the Configuration-Wifi.
Since I've only worked with .ino- and .h-Files within the Arduino System, I don't even know where to start debugging...
Everything so far seems to be booting fine according to the Serial Output
(Serial and Commandline output added as log.txt)

Can somebody help me to get the WiFi up and running?
Thanks in advance!!

[log.txt](https://github.com/softypit/esp32_mqtt_eq3/files/4276939/log.txt)

softypit, 2020-03-02:
Hi,
The log file shows the ESP is happy and reports it is running an AP.You should see an AP with the SSID 'HeatingController' appear in the 2.4GHz wifi range. This AP is open so you should be able to connect with no password and browse to 192.168.4.1 using your web browser.

Paul.

Wetterprophet, 2020-03-03:
Hey Paul, thanks for your reply!
That's what i thought, too. I tried out 4 different Sparkfun ESP32 Thing, but none of them were able to provide a working WiFi, neither did they find the BLE-eq3-Valve that I had close to them. I tried it with a cheap NodeMCU module with the ESP32 WROOM on it, which worked out of the box. I'll attach the log file. Do you have any idea, why it won't work with the Sparkfun module? I'd love to use this module since it has the ability to attach and charge a LiPo out of the box.

The biggest differences I've found so far:
Sparkfun ESP32 Thing:
Chip is ESP32D0WDQ6 (revision 0),
Features: WiFi, BT, Dual Core, Coding Scheme None,
Crystal is 26MHz

NodeMCU with WROOM32:
Chip is ESP32D0WDQ6 (revision 1),
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None,
Crystal is 40MHz

[log_working_ESP32WROOM.txt](https://github.com/softypit/esp32_mqtt_eq3/files/4281087/log_working_ESP32WROOM.txt)

Thanks for your help!

softypit, 2020-03-03:
Hi,
the sdkconfig was set for a 40MHz crystal which could be the problem.
I've rebuilt with the crystal frequency set to 'auto' which will hopefully work for both sparkfun and WROOM32 devices.
Can you try the new binary (1.56-beta prerelease) and let me know if it works.

Paul.