softypit/esp32_mqtt_eq3 #53: LED on ESP32

pmknowles, 2020-03-05:
Got mine working well with 5 EQ3s so thank you.
Not a programming issue but, nevertheless, would be useful. The LEDs on my ESP32s are painfully bright. It would be good to have a command to turn them off or only flash if there was an issue (loss of WifFi, MQTT or low battery on the EQ3).

softypit, 2020-03-05:
Which ESP32 module are you using?

pmknowles, 2020-03-05:
I think it's a Sparkfun one. It has a green and blue LED which will light the room at night!!

Regards

Phil

softypit, 2020-03-05:
The ESP32 boards I have seen use either gpio2 or gpio5 for an indicator LED. I believe the sparkfun boards also have an LED permanently connected to power (there may be a bridge/link you can remove to disable this).
This application uses GPIO2 for a status indicator as it is based around a generic WROOM32 devkit from ebay.
Check which GPIO your board uses for its second LED by looking up the schematic or hookup guide.

Regards.

Paul.