softypit/esp32_mqtt_eq3 #54: Idea: Add white and black list of eq3 devices

JuanGrados, 2020-03-15:
Hi, first of all, thanks a lot for this project, it is really nice!

JuanGrados, 2020-03-15:
Here is the code I am using in Home assistant in case it helps to understand my problem:

`
MQTT ID: eq3mqtt/
`

`
climate:
  - platform: mqtt
    name: Living room heating
    device:
      identifiers: '00:1A:22:XX:XX:XX'
    modes:
    - "off"
    - "auto"
    - "heat"
    precision: 0.5
    min_temp: 4.5
    max_temp: 29.5
    temp_step: 0.5
    mode_command_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/trv_mode"
    temperature_command_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/trv_temp"
    mode_state_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/status"
    mode_state_template: >-
      {% if value_json.mode == "manual" %}
        {% if value_json.temp == "4.5" %}
          off
        {% else %}
          heat
        {% endif %}
      {% elif value_json.mode == "auto" %}
        auto
      {% endif %}
    temperature_state_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/status"
    temperature_state_template: "{{value_json.temp}}"
    current_temperature_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/status"
    current_temperature_template: "{{value_json.temp}}"
    json_attributes_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/status"
    json_attributes_template: "{{ {'mac':value_json.trv, 'valve':value_json.valve, 'boost':value_json.boost, 'window':value_json.window, 'battery':value_json.battery} | tojson}}"
    away_mode_state_topic: "/eq3mqtt/00:1A:22:XX:XX:XX/status"
    away_mode_state_template: >-
      {% if value_json.mode == "holiday" %}
        on
      {% else %}
        off
      {% endif %}
`

`
automation:
  - alias: eq3mqtt_Temperature_Helper
    description: ''
    trigger:
    - platform: mqtt
      topic: /eq3mqtt/+/trv_temp
    condition: []
    action:
      service: mqtt.publish
      data_template:
        payload: >
          {% if trigger.payload == 4.5 %} {{trigger.topic.split('/')[2]}} off 
          {% else %}  {{trigger.topic.split('/')[2]}} settemp {{trigger.payload}}
          {% endif %}
        topic: /eq3mqtt/radin/trv

  - alias: eq3mqtt_Mode_Helper
    description: ''
    trigger:
    - platform: mqtt
      topic: /eq3mqtt/+/trv_mode
    condition: []
    action:
      service: mqtt.publish
      data_template:
        payload: >
          {% if trigger.payload == "heat" %} {{trigger.topic.split('/')[2]}} manual
          {% else %} {{trigger.topic.split('/')[2]}} {{trigger.payload}}
          {% endif %}
        topic: /eq3mqtt/radin/trv

  - alias: eq3mqtt_status
    trigger:
    - platform: mqtt
      topic: "/eq3mqtt/radout/status"
    action:
      service: mqtt.publish
      data_template:
        topic: "/eq3mqtt/{{ trigger.payload_json.trv }}/status"
        payload: "{{ trigger.payload_json | tojson }}"
`

As you can see, I need to duplicate the automations to handle two different esp32 devices.

pmknowles, 2020-10-10:
I have a couple of radiators where the same thing happens. I'm running openHAB and I look at the signal strength and just send the command to the one with the highest strength and ignore the other ESP32. for both send and receive.