softypit/esp32_mqtt_eq3 #55: Idea: Automatic continuous polling of Thermostat status and update of MQTT topic even if no command is issued.

elggem, 2020-10-01:
Hi, first of all thanks for all of your work and this code, it's working great over here! 🙏 

If I am not mistaken in the current version the thermostat status is only being polled and the MQTT topic updated after a command is issued. I would like to add an automatic polling feature (every minute or every 30 seconds) that updates the MQTT topic. This would allow me to monitor certain things like valve state over time and also allow feedback if the valve is controlled in the automatic mode or manually via the dial.

I can do some work and get familiar with the code, but I wanted to ask the maintainers first if there is any obvious reason why this isn't currently implemented or might not work as I think it will. Any other pointers also highly appreciated.

All the best from Europe!

softypit, 2020-10-03:
Hi,
there are a couple of problems with continually polling the TRVs.

Firstly there is no 'poll' command to get a status response from the valve without changing some setting. So in order to poll the valve we would need to repeatedly set 'auto' or 'manual', 'lock' or 'unlock' or the desired temperature. In the past I tried using 'unboost' as a poll command but there seems to be a bug (or perhaps some un-documented feature) in the TRV software when using unboost when the valve is not already in boost mode. The valve goes into a strange boost state without any change to the LCD display.  

The second problem is that continually polling the valve via bluetooth will likely affect the battery life. Each transfer requires the BLE radio to power up fully to exchange data whereas normally it will be in a low-power idle state.

My recommendation for repeatedly polling a valve would be to publish a command to set a value via MQTT each time you want to receive a status update. For example publishing a 'manual' mode command will repeatedly set the valve to manual mode and cause valve status to be published in response. 

Regards.

pmknowles, 2020-10-10:
I send an MQTT 'lock' command every 5 minutes to get the status. As the temperaure generally a is slow changing variable I suspect I could poll less frequently and not really lose functionality - I'm really turning radiators off in unoccupied rooms. Once the temperature is set the TRV does the job as it's supposed to.

krmichal, 2021-01-13:
Don't those TRVs make some broadcast with status? Like some Mijia BLE Thermometers, AFAIK? Then you would not have to actively poll the device, just wait for them to send periodical information.