softypit/esp32_mqtt_eq3 #58: Device unavailable

pyx1337, 2020-11-30:
I used it the complete last winter and now i wanted to get it to work again and cant control my temps...
but now i cant get it to work somehow...
esp setup:
[image]
[image]

My setup in nodered (businesslogic)
[image]
controlled via lovelace in home assistant (user interface)
[image]

Lordcybertracker, 2020-11-30:
Beim Mqtt Senden mußt du Mac klein schreiben.

pyx1337, 2020-11-30:
Same issue without mqtt via "Control EQ3 device"..

Selber Fehler wenn ich es ohne mein Gebastel rein über den interfacebutton steuere.

pyx1337, 2020-12-01:
[image]

This morning next try (no automations currently, only working with it on debugging purpose) it works, but 2 mins later same issue