softypit/esp32_mqtt_eq3 #59: Newbie help  + battery level

arj44ac, 2020-12-24:
Good day, im a newbie and i want to have a minimal control over BLE EQ3 TRV. Since my valves ar ein manual mode i just need to send 2 data of temperature (eg 25 and 10 deg) to switch them on and off basically. Plus i want to read the battery level.
Unfortunately i dont use MQTT and i wanna control those valves when im far from home so i wanna use some cloud based service. I know how to code the part reagarding online reading of desidered temperature. i just need a simple command to send to my EQ3 that temperature. Can u help? Plus (nice to have) how to read the battery status.
PS: i know the MAC of my valves.
thanks and marry xmass

larkingabout, 2020-12-25:
you can use a cloud based mqtt broker, and just use a script to send the relevant publish commands to the broker you select.
cloud based broker however involves many the problems/features of the oem app however. and some more..  

hosting your own broker if you dont run a 24/7/365 server at home is NEED NOT BE EXPENSIVE.  a £10 raspberry pi-0-w hosts both my mosquito mqtt broker and node red to control the 50 odd mqtt clients.. 

to get the battery level you just need to publish a command that generates a response that includes the battery level (unlock for example) and just subscribe to the OUT topic to recieve the data.  in fact the temp change command would also provoke publishing the battery metrics iirc, check it out in the web interface..

arj44ac, 2020-12-26:
Thanks for your answer and merry Xmas.  Unfort i cant have my own server with mosquitto coz i use a non propietary wifi , so i cant access to router(to open ports) and i think this will not allow me to send commands if im away from home (m i correct?). that why im working in cloud. I also want to avoid MQTT broker for the problems u mentioned, so my plan is to publish a value on Thingspeak private channel and time to time ESP32 will read it (i dont need fast reaction of my valves). So my script on ESP32 should be:
- read thingspeak every X minutes
- if value is HIGH, then send to EQ3 25 deg
- if value is LOW, then send to EQ3 11 deg
i know the MAC address, characteristics and services but untill now i was not able even to read a single charact. Can you please post a simple read and a simple write sketch? for c file of your project i was not able to extract.
Thanks a lot

larkingabout, 2020-12-28:
I think you have misunderstood the requirements to host a mqtt broker like
mosquitto.  while you may need to open a port to access your local broker
directly from outside your home. To use mqtt locally you do not need to
make any router mods.  You can still use any other service such as
thingsspeak as mentioned above to signal the request for action, but with a
locally hosted mqtt brpker all your responding script/system (hint node red
) need do is monitor the flag your using and publish the corresponding mqtt
topic/Payload to accomplish your desire.

a mqtt broker is not a heavyweight server system hogging resources an $1
esp8266 can host a broker serving a handful of clients.. and hosting a
broker locally will not expose you to any threats..

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

softypit, 2020-12-29:
esp32_mqtt_eq3 was written to be a server (more than one would be required in a typical home) controlled by some 'management' device like a raspberry PI or similar within the LAN.
The ESP 32 was never intended to have access to the internet and although Thingspeak allows unencrypted REST calls IMHO this is bad practice and all traffic from a LAN to the internet should be TLS encrypted at least. It would be possible to implement this on the ESP32 but configuration would be more complex.
I agree with @larkingabout that this system would be best achieved using something like a PIzeroW running mosquitto. The scripting to poll Thingspeak and send commands to the ESP32 would be relatively trivial using node/python/bash scripts.

If you only want to control the EQ-3 with thingspeak it would probably be simpler to use a bluetooth RPi (3B+/4/zeroW) and something like https://github.com/Heckie75/eQ-3-radiator-thermostat and use cron to run wget to fetch the thingspeak setting.