softypit/esp32_mqtt_eq3 #61: Wifi connection re-establish

d4fseeker, 2021-03-05:
When the WIFI connection is lost,  EQ3-MQTT seems to be unable to a working state and remains unreachable until physically rebooted.  I am on version EQ3-MQTT-ESP32 1.55-beta on a generic ESP32 Devkit.

Is there any possibility to have the code automatically retry WIFI or alternatively hard-reboot via Watchdog?

cichytom, 2021-03-05:
I have the same.
By the way, I have a question. Will there be a final (not beta) version?

Pasha1402, 2021-03-09:
It is not WiFi connection lost, but clash between WiFi and BLE on ESP32. Even soft reboot will not bring your device back to live - only hard reboot. There is (semi-)active discussion on this topic in parallel issue thread https://github.com/softypit/esp32_mqtt_eq3/issues/43

softypit, 2022-01-01:
It's taken me ages to find the time but I managed to build a new version of this application based on the latest esp-idf updates. I have no idea if this will make the WiFi issues magically disappear as I was unable to reproduce them on the previous versions.
I've been running this new version (v1.63) on my home-automation system for a few weeks now and it seems to work fine.

Regarding the 'beta' version - I intended to change it to remove the beta but I forgot with this one (v1.63). However there is no functional difference between this and a non-beta so please consider it a usable release.