softypit/esp32_mqtt_eq3 #65: EQ-3 notify error

giuseppedix, 2021-12-12:
Hi, Thanks for the excellent code. I have 9 TRV polling every 5 mins with command settime. 2 of them respond always with this error message: EQ-3 notify error.
Any suggestion? 
Thank you very much

softypit, 2022-01-01:
Hi,
This error is due to the notification registration within the BLE client being rejected. This isn't something I've seen before. Is there anything different about the 2 EQ-3s that exhibit this behaviour? i.e. are they newer models or running different software?

If you are able to monitor the serial output of the ESP32 it will display an error 'REG FOR NOTIFY failed: error status = x'. The status value may help to narrow-down the issue.
Regards.

giuseppedix, 2022-01-02:
Hi,
Thank you for the reply. It doesn't happen always to the same two TRV. Rebooting ESP32 the error is shown on 2 TRV different from the previous. With 7 TRV everything works perfectly.
Giuseppe

giuseppedix, 2022-01-02:
In the next days I will try to monitor the serial output.
Thanks

giuseppedix, 2022-01-10:
Hi, this is the error from the serial:
17:43:07.604 -> 0;31mE (874582) BT_APPL: Max Notification Reached, registration failed.
17:43:07.604 -> 0;31mE (874582) EQ3_MAIN: REG FOR NOTIFY failed: error status = 128
17:43:07.604 -> 0;31mE (874592) EQ3_MAIN: Command failed - requeue for retry

softypit, 2022-01-16:
Looks like a misunderstanding on my part with the notification registration. It seems the registrations are persistent even after disconnection from the EQ-3 device. There is a limit in the BLE stack virtual server of 7 registered notifications.
I've added an unregister step in the valve control process. I haven't been able to test this as I don't have enough EQ-3s within range of one esp32.
There's a new release (1.64) that contains the change. Let me know if this fixes the issue.

giuseppedix, 2022-01-31:
Yes, it does! thank you

closed 2022-01-31 by giuseppedix