softypit/esp32_mqtt_eq3 #67: New Firmware Version introduces real pairing

lennartjuette, 2022-01-23:
Just today I installed a new version of the firmware: v1.46. It looks like the device now uses a new PIN with 6 digits instead of four and the pairing process has been implemented properly. At least I actually had to enter the code when pairing with a RaspberryPi, while devices with v1.20 paired without a code.

Will proper pairing be implemented or do we have to avoid updating the firmware on our devices?

softypit, 2022-01-26:
I've upgraded one of my EQ-3s to 1.46 today and it still seems to communicate fine with this application. I can confirm that when using calorBT on an android phone I need to use the 6-digit pairing code with this device.

Can you check that your device still works too? If so I guess we can assume that this application will continue to work with this firmware.

lennartjuette, 2022-01-28:
To be honest I haven't even started working with your project, yet. I just stumbled upon this repository and didn't find any info about pairing, or that pairing is not required. So I was just wondering if this is still true.

I did some tests on a RPi with bluetoothctl and was actually forced to enter a code for the single upgraded thermostat, whereas the outdated ones could still pair without any interaction/PIN.

If I understand you correctly, you tested with an already paired device, right? In my case there's nothing set up, yet. Since I have a working solution already, I'm not sure when I can spare the time to test it. If you are ok with keeping this issue open for a while, I'll report back as soon as I find the time.