softypit/esp32_mqtt_eq3 #69: Undocumented Wifi and no HTML

Bonscha, 2022-04-04:
Hi,

I just flashed these images on Windows 11 - via ESP32 download tool v3.9.2 and having the proper driver installed - onto an ESP32-WROOM-32 that I had lying around.

I get a Wifi called "ESP_CC1995" instead of the documented Wifi name. Also it doesn't require a password. When connected, I don't get any HTML page under "192.168.4.1".

What could be the problem?

softypit, 2022-04-05:
My guess is that the download didn't succeed. The AP name looks like a default that would be used by many applications (but not esp32_mqtt_eq3). 
You should be able to confirm this by looking at the serial output when the device boots.

Bonscha, 2022-04-06:
Hi, great to hear from you so quickly! I have checked that yesterday, but didn't copy the serial output. It seemed there was no bootloader found, so the ESP32 couldn't even boot.

I erased via download tool, then flashed all 4 .img files on the respective memory adresses as indicated by the console script - but via Win GUI. Did that several times with two different and new devices - and didn't even get a Wifi AP anymore.

Does this even work with a ESP32-WROOM-32? I don't really find any other version and I'm not too familiar with the product family and potential differences in memory/firmware compatibility. If another variant is needed, I'll order that...

Bonscha, 2022-04-06:
Here is the serial output, which loops with a high frequency:

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
`

softypit, 2022-04-06:
That definitely looks like a programming issue. I use ESP32-WROOM-32s with this application so it should work with yours.
I don't run windows so have not used the ESP32 download tool.
Are you able to try with the python tool (esptool.py)? If you can run that it should work with the flash command in the readme.

softypit, 2022-04-08:
I'm not sure what else to suggest. 
There's https://github.com/hanhdt/esp32-flash-tool which could be worth a try. Otherwise if you have a raspberry pi you could use that as a programmer with the linux tools.

Bonscha, 2022-04-15:
Should have checked these boxes in ESP32 Download Tool before flashing, I guess...
I could facepalm myself all day long!

[image]

closed 2022-04-15 by Bonscha