softypit/esp32_mqtt_eq3 #70: "Not connected" - How to configure?

Bonscha, 2022-04-15:
After initializing and the ESP32 finding the test TRV I have installed, I have problems connecting it to my Openhab 3 installation.

Raspi with Openhab has IP 192.168.178.72 (or "openhabian")
ESP32 has IP 192.168.178.83
Mosquitto is installed on Raspi, MQTT broker in OH3 is connected w/o SSL to Mosquitto.
What do I have to enter in the ESP32 config to get it working with my MQTT broker in the Raspi?

[image]
[image]
[image]

I already tried with:

192.168.178.72
192.168.178.72:8883
192.168.178.72:1883
mqtt://192.168.178.72
mqtt://192.168.178.72:8883
mqtt://192.168.178.72:1883
openhabian
mqtt://openhabian

Bonscha, 2022-04-15:
After I somehow managed to understand it needs JSON conversion to put/read strings to/from the correct payload slots, I just need help with a correct JSON conversion template for OH3. Why is this not documented anywhere?

If someone can give me an example so I can work it out, I'll write an extensive wiki entry about OH3 integration... -.-

softypit, 2022-04-15:
It is important to understand the way the valves operate with regard to BLE signalling. This is basically the way the mqtt command/status mechanism works.
i.e. you send a command like set-temperature or boost and the trv sends back a response with the full current status of all parameters. This obviously doesn't map directly to OH items so some translation work is required.
The json structure is defined in the readme.md file. Each of the parameters could be mapped directly to an OH item provided the facility exists within OH3 to achieve this from a json object.
You can also confirm the exact json status payload returned after each command by looking at the status log on the ESP32 gui.

To send a command you need to publish to the command topic with a message payload containing the BLE address of the valve and the command and any parameters.
topic is '/<mqttid>radin/trv' where mqttid is the id configured on the ESP32. 
The message would be '00:1A:22:1A:85:BE settemp 22.0' to set the temperature to 22.0 degrees.

To do this from the command line as a test use:
mosquitto_pub -h <broker address> -p 1883 -t "<mqttid>radin/trv" -m "00:1A:22:1A:85:BE settemp 22.0"

Bonscha, 2022-04-16:
Thanks for clarification this far.

Before, I tried with:
`transformationPatternOut: JSONPATH:$.[?(trv=='00:1A:22:1A:85:BE')].settemp`
Which didn't do anything...

I understand that I don't have to use JSONpath conversion for sending - I can just use "formatBeforePublish" and it sends a command which is also received correctly by the device:
`formatBeforePublish: 00:1A:22:1A:85:BE settemp %s`

After getting the control to work, I rewrote the JSONpath for incoming telegrams to:
`transformationPattern: JSONPATH:$.temp`
This works and I get the temp feedback. Since now the MAC address is missing, though, I'd suspect when having several devices online, the values won't be able to get matched to the correct points in OH3...?

I don't seem able to "pick" the ID-related values only, though - my transformation pattern including the ID (see above) produces errors, even though it's valid in the online evaluator: http://jsonpath.com/

softypit, 2022-04-16:
I've not worked with JSONpath before (or OH3) but I have played with OH2 rules - would something like the following work as a rule? 
you may need to manipulate the 'valve' string to remove the colons ':' if OH3 doesn't like them in an item name

rule "Handle EQ3 status"
  when
    Item EQ3_status_json changed
 then
    val valve = transform("JSONPATH", "$.trv", EQ3_status_json.state.toString)
    val temp = transform("JSONPATH", "$.temp", EQ3_status_json.state.toString)

    // Assume temperature setting item for this valve is "00:1A:22:1A:84:05-settemp"
    postUpdate(valve+"-settemp",temp)
    // Update any other items associated with this trv here
    val mode = transform("JSONPATH", "$.mode", EQ3_status_json.state.toString)
    postUpdate(valve+"-mode",mode)
 end

Bonscha, 2022-04-17:
Thanks, that did the trick! I adapted the script a little, since OH3 doesn't let me name the items with ":", so I cannot use the MAC addresses, but no problem. I do parse the whole string and send updates specifically to all existing items. This way, It does work fine and gets matched correctly to the respective device's items.

[image]

Here is my parsing script (DSL script):
`  // parse the whole status string from listening channel

val trv = transform("JSONPATH", "$.trv", EQ3Bridge_EQ3Status.state.toString)
val temp = transform("JSONPATH", "$.temp", EQ3Bridge_EQ3Status.state.toString)
val offsettemp = transform("JSONPATH", "$.offsettemp", EQ3Bridge_EQ3Status.state.toString)
val valve = transform("JSONPATH", "$.valve", EQ3Bridge_EQ3Status.state.toString)
val mode = transform("JSONPATH", "$.mode", EQ3Bridge_EQ3Status.state.toString)
val boost = transform("JSONPATH", "$.boost", EQ3Bridge_EQ3Status.state.toString)
val window = transform("JSONPATH", "$.window", EQ3Bridge_EQ3Status.state.toString) 
val state = transform("JSONPATH", "$.state", EQ3Bridge_EQ3Status.state.toString)
val battery = transform("JSONPATH", "$.battery", EQ3Bridge_EQ3Status.state.toString)

  // match trv MAC address to respective OH3 items and send updates
  
      //Schlafzimmer
if (trv == "00:1A:22:1A:85:BE") {
    postUpdate("Heizung_Schlafzimmer_Solltemperatur_Status",temp)
    postUpdate("Heizung_Schlafzimmer_Mode_Status",mode)
    postUpdate("Heizung_Schlafzimmer_Ventilstellung_Status",valve)
      //Wohnzimmer
} else if (trv == "00:1A:22:1A:84:05") {
    postUpdate("Heizung_Wohnzimmer_Solltemperatur_Status",temp)
    postUpdate("Heizung_Wohnzimmer_Mode_Status",mode)
    postUpdate("Heizung_Wohnzimmer_Ventilstellung_Status",valve)
}`

Since I'm on vacation the next week, I will write an extensive wiki entry on how to implement in Openhab 3, including flashing ESP32 for dummies... Thanks for the help so far. I'd leave this issue open until documented in the wiki, if that is okay for you. :)

Bonscha, 2022-04-17:
First draft for wiki entries is ready here: https://docs.google.com/document/d/1Hkb8LttefBZPBlLxWwad50e2exZTauphOs8B9PFYjnc/edit?usp=sharing

Feel free to use that for this wiki or anywhere else, to prevent questions like mine in the future... ;)